hultenvp/solis-sensor #442: Solis API with Control - only on first inverter

Tadpole48, 2025-04-04:
I selected to have the Beta Solis Control, this works but I have two Solis inverters, the first in the list has "Controls" AND "Sensors", the second in the list only has "Sensors" (the second inverter is a hybrid with batteries and that is the one I really need control over)

Versions
HA OS Version 15.1
HACS version 2.0.5
Integration version 3.9.0

I am VERY new to HA and the Solis API - please be kind! (PS. I duplicated this report with fboundy)

skrachs, 2025-04-08:
The same situation. I have two inverters in parallel.
But I see controls only for the first one.

Also the one who has controls is a Slave one.
Don't see controls for a Master.

mjatkey, 2025-05-03:
Guys, I think this may be resolved, I have two inverters and have charge/discharge parameters working on both at the moment without any issues.

Tadpole48, 2025-05-04:
Not for me, Just reinstalled and all is the same, controls are still on the wrong inverter.  So I have switched integration to the Modbus version and used some Waveshare RS485 to TCP adapters and install MQTT broker, all works great now, 5 second updates and full control of hybrid inverter.

japers, 2025-05-20:
I have the same issue - I can only control the first inverter. Would be great to be able to also control the second inverter.
(I have Modbus control but I have found this to be unreliable with two inverters)

Tadpole48, 2025-05-20:
> I have the same issue - I can only control the first inverter. Would be great to be able to also control the second inverter. (I have Modbus control but I have found this to be unreliable with two inverters)

I use MQTT on the inverter with battery and a Shelly Power Monitor on the other inverter (the Shelly also providing MQTT data)

TCW82, 2025-05-25:
Just to add some more data to the pile.. I have issues too. I did quite a few things already, among which: I separated the 2 inverters into two separate Solis (cloud) plants. This did nothing towards a solution. Both plants have on off control in the solis cloud app (as they did before). 
@mjatkey what was your path to solution in this?

TCW82, 2025-08-11:
I linked this to my most recent bug. After the troubles earlier in July, I now have 1 inverter that allows me to control the device in h.a. and 1 inverter that does not allow me to control it in h.a.