hultenvp/solis-sensor #443: Energy Storage Control Switch Status

stevemel56, 2025-04-05:
I would like to see the numeric status of this exposed as it used to be in the Solis App.
This would enable conditional switching only if needed bearing in mind there are multiple ways this switch can be set not just a single HA instance.
I also find that the switching is not reliable and does fail. A status would enable this to be retried until successful.

Not a great fan of the text for options and would like a simple numeric alternative or the ability to set specific bits  but understand why it is like this.

mkuthan, 2025-04-05:
I had a real battle over the last 2 weeks with Solis Cloud Control API.

1. Not every bit combination is correct. In the best case, it will be ignored; in the worst case, it could damage the inverter. I prefer not to take that risk.
2. The Solis Cloud Control API is flaky, and response handling is non-trivial.
3. I haven't found a better way to model Storage Mode than using an option list. The list of options should be short (Self-Use, In-Feed Priority, Off-Grid), but supplementary options might be exposed as entity attributes. For example, Backup Mode or Allow Grid Export.

If you have the time and interest, you are welcome to try my experimental integration. I am continuously working on improving it, but it currently works very stably.

You can find it here: [Solis Cloud Control Integration](https://github.com/mkuthan/solis-cloud-control)

thybag, 2025-04-06:
Was about to raise a similar bug - my "Solis Energy Storage Control Switch" status appears to just read as `unknown`. I wasn't sure if this was intentional, or just a difference in the specific bit combinations its looking for?

Be nice if it could return the current status, so my automation can both determine if it needs to change or confirm its updated as expected.

Manual testing (looking at the solis control website to see if expected changes are reflected) seems to show the changes come across okay.

stevemel56, 2025-04-06:
I understand some bit combinations may not be good but I would also expect the inverter to validate the value supplied - maybe that’s a big ask though. I’m happy with the combinations there are as I only use 2 and can only envisage using a couple more. The ability to view what the inverter has is pretty important though. I’ll have a look at the experimental and see if that does what I need.

lazy-pete, 2025-06-29:
I seem to stall fairly regularly with an error of “retrieved invalid state of unknown for select.solis_energy_storage_control_switch” 

Is this remediable ?

lazy-pete, 2025-06-30:
Hi - in case it is of help to anyone, I seem to have resolved this particular issue of the PV-opt automation erroring for an unknown energy storage control switch status, by swapping it to use the Inverter Control Storage Mode entity provided by the separate Solis Cloud Control integration listed above. That seems to report a status reliably.

stevebuk1, 2025-08-14:
In control_const.py, its because the ALL_CONTROLS = False (i.e. 3 slot option) lists this for the energy storage control switch:

        "636": [
            SolisSelectEntityDescription(
                name="Energy Storage Control Switch",
                key="energy_storage_control_switch",
                option_dict={
                    1: "Self-Use - No Grid Charging",
                    3: "Timed Charge/Discharge - No Grid Charging",
                    17: "Backup/Reserve - No Grid Charging",
                    33: "Self-Use - No Timed Charge/Discharge",
                    35: "Self-Use",
                    37: "Off-Grid Mode",
                    41: "Battery Awaken",
                    43: "Battery Awaken + Timed Charge/Discharge",
                    49: "Backup/Reserve - No Timed Charge/Discharge",
                    51: "Backup/Reserve",
                    64: "Feed-in priority - No Grid Charging",
                    96: "Feed-in priority - No Timed Charge/Discharge",
                    98: "Feed-in priority",
                },
                icon="mdi:dip-switch",
wheras the 6 slot option lists this:

        "636": [
            SolisSelectEntityDescription(
                name="Energy Storage Control Switch",
                key="energy_storage_control_switch",
                option_dict={
                    "33": "Self-Use Mode - Allow Grid Charging",
                    "49": "Self-Use Mode - Allow Grid Charging, Backup Mode On",
                    "1": "Self-Use Mode - No Grid Charging",
                    "17": "Self-Use Mode - No Grid Charging, Backup Mode On",
                    "96": "Feed-In Priority Mode - Allow Grid Charging",
                    "112": "Feed-In Priority Mode - Allow Grid Charging, Backup Mode On",
                    "64": "Feed-In Priority Mode - No Grid Charging",
                    "80": "Feed-In Priority Mode - No Grid Charging, Backup Mode On",
                    "5": "Off-Grid Mode",
                },
                icon="mdi:dip-switch",
            )

i.e. the key value is in quotes. Adding quotes to the values has solved the issue on my system.