hultenvp/solis-sensor #458: How to set up the API control

hannes3510, 2025-05-08:
I don't understand how i can set up my converter control?

i got the key and secret. Any one got a tutorial?

plebcity, 2025-05-08:
> I don't understand how i can set up my converter control?
> 
> i got the key and secret. Any one got a tutorial?

You need to insert the password to the portal, not the API secret. If you go to soliscloud -> select your inverter -> Top right click inverter control, you have to insert a password there this is the password you need to insert in the configuration for the soliscloud integration in HA

hannes3510, 2025-05-08:
Thank you for the reply,

i enabled the check box, used my account password as portal password, press submit, but nothing happens? I rebooted HA , still the same configuration options.

crryp, 2025-05-08:
Might be an issue with it. This has been working for my inverter in the past. When i do try to setup. all values will not be initiaziled. Whenever i put the control off again i can read the values again.

jmason, 2025-05-08:
I think the same configuration options will appear on the "Configure" page -- but once it's been filled out, and after a restart, you should see the Controls on the Device page:

[image]

AleLorenzi, 2025-05-09:
I also didn't get any feedback from the checkbox, so I thought I was using the wrong password...the right one is the Sols portal password. It seems to be unchecked, however after restarting the integration the configuration section will appear

eviltabouleh, 2025-05-11:
My controls modal appeared after entering the Soliscloud password and a HA restart. Thanks guys this is an awesome addition

hannes3510, 2025-05-11:
hmm when i try it, my connection seems to stop with soliscloud. I no longer get my values from my inverter.

crryp, 2025-05-11:
> hmm when i try it, my connection seems to stop with soliscloud. I no longer get my values from my inverter. 

Yes thats the same for me! But also when i tried the normal interface i dont get values when i read (the control values) so might be solis related? Can you try to see any value in the solis app? On/off for example. It shows me command successfull but i never can read the value. And i am sure that worked before.

crryp, 2025-05-14:
@chrisjglen does the normal interface in the app or soliscloud give you any return values?
For me it sent the read commando but never returns a value. To see if this might be a solis related issue can you try?

crryp, 2025-05-14:
Not at home right now but for me the values never return on the solis app or cloud (they did in the past) for you it seems to work. Maybe its because of my dns settings since i changed them recently. 

Inverter: S6-GR1P4k
Using a datalogger with cable not wifi.

Not sure why HA doesnt do anything that also worked in the past.

crryp, 2025-05-14:
Just made a solis ticket since im stuck here and that might also be the reason the ha intergration gets a timeout. 

Its dutch but stated: commando send, no data results.

[image]

[image]

plebcity, 2025-05-14:
> Just made a solis ticket since im stuck here and that might also be the reason the ha intergration gets a timeout.
> 
> Its dutch but stated: commando send, no data results.

This is because you don't have a S3-WIFI-ST datalogger or better

crryp, 2025-05-14:
> > Just made a solis ticket since im stuck here and that might also be the reason the ha intergration gets a timeout.
> > 
> > Its dutch but stated: commando send, no data results.
> 
> This is because you don't have a S3-WIFI-ST datalogger or better
> 
> 

It worked half year ago perfectly fine in the solis app? So they left support for mine or?

plebcity, 2025-05-14:
> It worked half year ago perfectly fine in the solis app? So they left support for mine or?
> 
> My datalogger is with lan cable and i did control it so guess thats not the case.

I don't know, What I do know is that the control doesn't work if you can't read the value, with my old data logger it showed -- aswell but with my new it shows either ON or OFF

crryp, 2025-05-14:
> I don't know, What I do know is that the control doesn't work if you can't read the value, with my old data logger it showed -- aswell but with my new it shows either ON or OFF

I get that but as i said it worked before so. Turned it off and on with the app. So they might have changed something not sure. Normaal the read result was there in a few seconds. So will see what they come up with on my ticket.

WimDeceuninck, 2025-06-09:
Hi @crryp, 
Hi @hultenvp, 

Do you have an update on the issue above? 

Since yesterday, I have not been receiving any data via my S3-WIFI-ST datalogger. 
I believe it may be defective, as I am unable to connect it to my router. 
To ensure continued data availability in Solis Cloud, I have replaced the S3-WIFI-ST datalogger with an older DSL-W datalogger, which naturally means the Solis sensor integration no longer works.

I believe I will need to purchase a new datalogger, and I am wondering which one I should choose. 
Should I opt for the S3-WIFI-ST again, or would the S4-WIFI-ST (using the 4P to USB adapter) also be compatible?

Inverter: Solis, Model: F5,  Machine Type: RHI-3.6K-48ES-5G

Thank you in advance for your input and assistance.

Kind regards,
Wim

crryp, 2025-06-26:
I got it working rn somehow the app of solis doenst show results. Anyway my own c# project im able to change values. Also with the intergation which i prefere to use since then i can use automations easier. 

But yesterday i found that the request limit on api v2 is 200 times a day. 

Im not sure what the polling rate of the controls is which is 5 min on the sensors. If thats also the case on the controls you will get to the limit quickly. Since its a loop for all control values.

Im not that used to python so didnt find yet where i can see the control read interval. I did remove all controls i dont use i only use one. Would be wonderfull if i can change the refresh raid to ones per hour so im never hitting that 200 request limit per day