hultenvp/solis-sensor #462: Feature Request

janusr, 2025-05-11:
Love this integration. :-D
Possibility to modify Feed In Limit would be of special interest as I sometimes have negative export price:

[image]

patduckless, 2025-05-12:
I think this is added by [#463](https://github.com/hultenvp/solis-sensor/pull/463) and released with v3.11.0

janusr, 2025-05-12:
Not quite same result. I don't want to limit the inverter output, but the export to the grid. Internal power use fluctuates, so setting the limit on the inverter is not good.

janusr, 2025-05-14:
Tested this, it seems to read the value but does not take the Hectowatt (factor 100) into account:
I think the odd reason is that it must be in 100w increments.
[image]

patduckless, 2025-05-14:
Yes it is a pain. Home assistant doesn't have hectowatt as a unit.

janusr, 2025-05-14:
Workaround: A helper and an Automation

Input Number Helper:
[image]

Automation:
```
alias: Update Solis Feed Limit
description: Updates solis_feed_in_power_limit based on master input in watts
triggers:
  - trigger: state
    entity_id:
      - input_number.solar_current_feed_in_limit
conditions: []
actions:
  - target:
      entity_id: number.solis_feed_in_power_limit
    data:
      value: >
        {{ (states('input_number.solar_current_feed_in_limit') | float / 100) |
        round(0) }}
    action: number.set_value
    enabled: true
  - target:
      entity_id: number.solis_feed_in_power_limit
    data:
      value: "124"
    action: number.set_value
    enabled: false
mode: single
```

janusr, 2025-05-14:
Side Note:
The maximum value needs to be read from the inverter.
I have 2 10Kw inverters - Thus maximum 24Kw (in the screenshot)
This depends on the Inverter(s) I'm not sure if its possible to retrieve this information, nor using it in the sensor.

closed 2025-06-26 by patduckless