hultenvp/solis-sensor #464: Using remote dispatch for custom control

levexis, 2025-05-12:
A Solis engineer warned me about how I was controlling the inverter. This is with the API schedules 1 & 2 both charge and discharge. He said we should be using the remote dispatch registers as the ones we are using are saved in flash memory (not ram) and limited to 10k cycles. 

"I was only concerned if you were changing the settings via direct modbus using the first time-charging registers.
With these "remote dispatch" registers you can actively develop your own control if you like"

I've attached the screenshot of his docs he shared with me. These sensors may be labelled differently in the API. The ones I am using are "Solis S6-EH3P Solis Grid Time of Use Charge Start (Slot 1)" etc.

I actually want to set these hour by hour, or possibly 4 times an hour from June (NL is moving to 15 min intervals). I'm hoping what he's said to me makes more sense to you and is helpful. If I can do anything to help then let me know, else please tell me which sensors I should be using :)

[image]

patduckless, 2025-05-12:
Could you share the entire document with us?

mjatkey, 2025-06-29:
Guys, I think this is real issue, so I have two Solis 5K hybrids in different locations/setups, and have been using the integration in HA together with a bit of automation to set the discharge current based on the soc of the battery and the length of time before the charge window opens.
So first inverter failed 2 weeks ago with OV-ILLC(1052) alarm, tried all the normal firmware update restarts etc but now it will only pass the PV power to grid. This one is being swapped free of charge by the installation company and was just up looking at the second Solis and that has failed with a DC-INTF alarm, I've not logged that with Solis yet as I want to get the first one swapped.
Now one failure maybe but two within a few weeks has to be connected to the whole API scripting bit.
I for one am going to disable is in HA for time being.