hultenvp/solis-sensor #468: Solis Energy Today sensor reduced precision

paulmthomas2, 2025-05-15:
I noticed today that the "Solis Energy Today" sensor - ie the one that updates with the solar yield so far each day, has, since the morning of 15 May 2025 stopped reporting in increments of 0.1kWh, and now only reports in full kWh. This would appear to be a change at the Solis end - so wondering whether anyone else has this - as my SolisCloud is also only displaying yield in full kWh increments - it showed nil this morning, until ticking over to 1kWh after a prolonged period. This shows as "Today Yield" in soliscloud. Interestingly on Solis Cloud in the inverter details view, whilst "Today Yield" still only increments in full kWh, on the chart I can see the full data for yield at 0.1kWh increments. Clearly the current API endpoint is the "Today Yield" rather than wherever the data for that chart comes from, but it is reassuring that the inverter is still providing that data.

So in terms of an issue, if Solis have permanently changed the rounding on that datapoint (I've emailed support to ask), then is it possible to change the endpoint to pull the more detailed data that the inverter details chart uses, as it's still clearly coming out from the datalogger?

Anyone else also seeing this, or is it just me?

THanks for all you do on this integration, which is brilliant, by the way!

patduckless, 2025-05-17:
Mine is still reporting in 0.1 increments. 
Do you have control enabled?
[image]

paulmthomas2, 2025-05-20:
I think this was Solis Issue - I reached out to them, but the next day it had returned to 0.1 increments.