hultenvp/solis-sensor #471: Newby setup issue

djdarlo, 2025-05-17:
This is my first time trying to enable the integration.  

1) I select soliscloud as the PV version (PV portal as https://www.soliscloud.com:13333)
2) Portal username is my email address which I use to logon to soliscloud
3) API Secret copied from the API page on the portal (after enabling)
4) API Key copied from the API page on the portal
5) Station ID copied from url (19 digit code)

However I get the following error message.

Cannot login with provided URL and credentials

Am I missing anything obvious?

patduckless, 2025-05-17:
Strange, could you try re-downloading the integrator from HACS?

patduckless, 2025-06-26:
@JimWuerch No there is not. There is debug logging within home assistant. Could you both try enabling that and posting the results here please? 
@djdarlo

hultenvp, 2025-08-22:
@djdarlo Was your issue similar to @JimWuerch ?

krizzziz, 2025-09-27:
@djdarlo did you enter you password at the bottom? I thought the password at the bottom was only for the "experimental" control feature, but you need to add it to be able to add the configuration. I didnt see the step of the PW in your list.

Dfg11333, 2026-11-03:
Had the same issue. Turns out, I used the station name instead of the ID (in the URL, 19 digits number, see Readme).

The error message is pretty cryptic, but I'm not sure it's easy to improve it. What could be done though, is adding a `(19 digits, see README)` in the field name.

And yes, I know this was documented, but I had installed Solis Cloud Control before, which doesn't ask for a username/email, and even lists the different stations _by name_ without a need to find the 19 digits ID, so the setup of this integration was much less straightforward and therefore a bit confusing.