hultenvp/solis-sensor #473: Controls affecting the SOC graph on Solis Cloud

DMonkey-hub, 2025-05-21:
I had this problem when the new controls were first added to the integration. It has been ok (I think) but has now returned. It's as if the controls are using the Solis API too often and produces the graph shown below with SOC dipping to zero frequently. It's fine when I disable the controls. It would be good to have read and write buttons to only use the API when required (assuming that's the issue).

[image]

**Versions**
Core 2025.5.2
Supervisor 2025.05.1
Operating System 15.2
Frontend 20250516.0

jmason, 2025-05-22:
Hmm, that's an interesting idea.

What we could potentially do is use a `refresh_interval` which polls the current values from the control API much less frequently, and then allow https://www.home-assistant.io/integrations/homeassistant/#action-homeassistantupdate_entity to be used as a button to refresh the current values from the API if needed?

DMonkey-hub, 2025-05-22:
Now that sounds like a plan :-)

Would be great if that fix it. I'm happy to do any testing if needed.