hultenvp/solis-sensor #474: only 1 of 2 inverters can be controlles in HA, both can be controlled in Solis Cloud App

TCW82, 2025-05-22:
**Describe the bug**
I own 2 inverters, using the same make/model WIFI data stick and both can be controlled in the solis cloud app. Only 1 inverter can be controlled in the HA intergration. 

All I would like to be able to do is switch on and off using HA based on dynamic energy prices. I can do this manually in the solis cloud app, but I seem to have a (probably minor) issue. In my setup, I run 2 inverters. The have the same (make / model) data stick.
Using the solis cloud app, I can control both inverters. I can swich them on and off, I van set the output % between 0 and 110%.
I have done the intergration in HA and there both inverters show up. The problem is that only 1 of these two is providing entities that allow control.
These both show 5161 (this: [https://github.com/hultenvp/solis-sensor/issues/440#issuecomment-2863483673].
I use the 3.11 version of the intergration.

patduckless, 2025-05-22:
Do they have different serial numbers in the web app? Can they be setup as two inverter entities?

TCW82, 2025-05-22:
Well, that is a good question. I think this is should be possilbe, but the way it is all setup (for some years already now) seems to work of the basis that there is 1 plant, 2 inverters and they are added based on the plant ID. 

What I did is this: To add the plant to HA I used the route advised on github, obtaining the 13 digit plant id. 
The data loggers, the inverters and the plant all have different (SN)-numbers.  I actually tried (more by mistake) to add the inverter using its unique sn-number, this was not possible in HA, I guess because the key provided by Solis is linked to the 13 digit plant id. 

Could you indicate what specific info I can provide in this matter?

skrachs, 2025-05-23:
Isn't this the same issue?
https://github.com/hultenvp/solis-sensor/issues/442

TCW82, 2025-05-23:
I had a Quick Look, it does seem so. Although there seem to be quite a few variations mentioned in the listed issue. 
I won't mind merging the issues, which seems sensible if they are indeed the same.

TCW82, 2025-05-25:
Ok, Today I took some time to separate the 2 inverters into their own separate "plant". This meant I also had to request a new API key for the new plant. Which I did receive. This allowed me to integrate the 2 plants separately in Home assistant. After the integration, the 2 plants had the exact same number of entities (188) and there was still no way to control the troublesome inverter.

TCW82, 2025-05-25:
> Do they have different serial numbers in the web app? Can they be setup as two inverter entities?

I think I did what you suggest here. I made 2 separate plants, and added them separately to HA, this was not successful.

TCW82, 2025-05-25:
Somehow this morning both inverters showed the desired control entities. It now seems to work.

closed 2025-05-25 by TCW82