hultenvp/solis-sensor #481: Integration fails to load

TCW82, 2025-07-12:
This is my first report, there might be (a lot of) room for improvement....

**Describe the bug**
In the current ha version, Solis inverters do not load anything anymore.

**Config**
Standard config from the repo

**Versions**
Installation method
Home Assistant OS
Core
2025.7.1
Supervisor
2025.07.1
Operating System
16.0
Frontend
20250702.1

Integration Version 3.12.0

**Debug logs**
Logger: homeassistant.config_entries
Source: config_entries.py:749
First occurred: 09:53:25 (3 occurrences)
Last logged: 09:59:48

Error setting up entry None for solis
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 749, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis/__init__.py", line 121, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(entry, CONTROL_PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2582, in async_forward_entry_setups
    await self._async_forward_entry_setups_locked(entry, platforms)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2593, in _async_forward_entry_setups_locked
    await asyncio.gather(
    ...<11 lines>...
    )
asyncio.exceptions.CancelledError

And perhaps this: 

This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/solis/service.py:191
integration: Solis Inverter (documentation, issues)
First occurred: 11 July 2025 at 16:48:10 (5 occurrences)
Last logged: 09:59:25

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 163, in async_discover
    await self._discover_controls(inverter_serials)
  File "/config/custom_components/solis/service.py", line 191, in _discover_controls
    control_desciptions = ALL_CONTROLS[hmi_flag]
                          ~~~~~~~~~~~~^^^^^^^^^^
KeyError: None

TCW82, 2025-07-12:
This seems to have started yesterday, end of the day. 

[image]

voogel, 2025-07-13:
Removing and re adding the inverter worked for me until I tried to enable the write api. I’m running the latest version of HA Core.

imc67, 2025-07-13:
same as: https://github.com/hultenvp/solis-sensor/issues/438 ?

TCW82, 2025-07-13:
> same as: [#438](https://github.com/hultenvp/solis-sensor/issues/438) ?

Possibly, although mine has worked flawlessly for a couple of months. It feels to me there is a strong link to the ha updates, but have insufficient knowledge of the deeper workings to be sure. 

I will try some of the suggested solutions later today and report back, although I am not hopeful.

TCW82, 2025-07-14:
Neighbour reported that his system is now working again. I do not have the remote access needed to check, in about 5 hrs. I will report back on the situation. But it might be solved.

voogel, 2025-07-14:
I can still set the Solis API control values of my inverter using an old script I created before HA had the power limit option so I don’t think at has anything to do with their API. For me all the issues started with the new version of HA Core.

imc67, 2025-07-15:
Some steps I took to reproduce and see the results:

1. I deleted the plant and with that the integration, restarted HA (just to be sure) and added the integration and plant with ALL credentials. 
2. I noticed that, in contrary to the screenshots in readme, in the initial setup you can add already the credentials for the Control API, I did. Though after setup there were no switches, only sensors.
3. After some minutes new data came in! But still no switches, only sensors
4. I restarted HA and again after some minutes new data came in! But still no switches, only sensors
5. Added "Solis Control" (again, because I did that in the initial re-setup), restarted HA
6. ⭕️ Now the integration is STUCK, no data, and with all aforementioned errors in the logs**

imc67, 2025-07-15:
> I can still set the Solis API control values of my inverter using an old script I created before HA had the power limit option 

Are you willing to share your "old script" so I/we can use that (temporarily) to keep using our automations to switch off or limit during negative energy prices?

J3G0, 2025-07-15:
Similar here. Suddenly no data at all.

Tried to reload, didn't work.

I removed the inverter, re-added it. The sensors came back but none of the controls are available for me. So no remote inverter switching. 

I also have the feeling it happened with a HA update.

TCW82, 2025-07-15:
I am going to submit a ticket to the Solis company aswell. Perhaps a two
way approach helps.

Did anyone here try putting an old version of HA back ?

sjanssen15, 2025-07-15:
Same for me, kept working until I updated to 2025.7.2. I don't think it's their API since it did work for me until 15-7

micdo54, 2025-07-15:
Update from me, received a answer from Solis. They asked me to try it again
and suddenly I could back login.
I received all data again. After one hour I update homeassitant to the
latest state. Every thing works still fine. I don’t use at the moment no
device controls in Solis sensor

sjanssen15, 2025-07-15:
with login, you mean remove the integration and setup the integration again? or in the app?

For me, reloading the integration or restart HA didnt fix the issue

TCW82, 2025-07-15:
I filled a ticket with Solis. Also deleted and reintergrated everything. 
That last step did nothing good.

plebcity, 2025-07-17:
My integration is still not working, it shows unavailable for all the entities and logging shows no errors other than it takes a long time to setup the solis integration

TCW82, 2025-07-17:
I am emailing with Solis support, but they are sending me into the woods
with API documents and postman api things. I can't make heads or tails of
them yet.

plebcity, 2025-07-18:
@TCW82 Could you please use that Postman collection to execute the inverterDetail request and report back the hmiVersionAll value you have? Also read the README file on how to setup the Postman environment with your credentials for Soliscloud.

plebcity, 2025-07-18:
> Hi, at the moment; no, for two reasons, one is geological in nature and the other is that I have no idea what you just wrote 😬😬. There might be some chance this afternoon, is it possible you help me along? (Could also be done with email or other direct form of communication?)
> 
> With the Solis people is am stuck at the "please sign the NDA so we can send you an API document"

You don't need to sign a NDA to get the API documentation, it is publicaly available here: https://oss.soliscloud.com/doc/SolisCloud%20Device%20Control%20API%20V2.0.pdf

That documentation doesn't help us since it doesn't say which cid's are available and which cid's arent't available. 

If you aren't a software developer then installing postman and setting it up is quite a hasle. What you could try however is doing inspect element on the soliscloud website. 

1. Go to the soliscloud website
2. Go to the inverter details page
3. Do inspect element
4. In the devtool window that opens go to network
5. Refresh the soliscloud page
6. Find the "detail" request
7. Go to the response and find the hmiVersionAll value

plebcity, 2025-07-18:
[image]

Got this reply from Solis when I asked what has changed on the response for CID 6798 which we use to determine HMI version. It is something totally different

TCW82, 2025-07-18:
> You don't need to sign a NDA to get the API documentation, it is publicaly available here: https://oss.soliscloud.com/doc/SolisCloud%20Device%20Control%20API%20V2.0.pdf
>
> That documentation doesn't [...]

I think there is time and i think my simple ass van do this. Thanks. Will get back to you asap

TCW82, 2025-07-18:
> You don't need to sign a NDA to get the API documentation, it is publicaly available here: https://oss.soliscloud.com/doc/SolisCloud%20Device%20Control%20API%20V2.0.pdf
>
> That documentation doesn't [...]

I know about the Api documentation. They asked me about the NDA the previous time I had issues with the h.a. integration. I asked the service person what the current version was. If it is >2.0 it might be interesting to obtain.

TCW82, 2025-07-18:
Ok, Perfect description, thanks. 
In the Response tab i found this (I XXX some details) 
the requested HMI: 
    "hmiVersionAll": "8015",

The code as a whole, with some triple X amendments. 
`{
    "success": true,
    "code": "0",
    "msg": "success",
    "data": {
        "chartAllParams": "u_pv1,i_pv1,pow1,mppt_upv1,mppt_ipv1,mppt_pow1,e_today,e_total,u_ac1,i_ac1,fac,pac,reactive_power,inverter_temperature,SIR_realtime",
        "id": "XXX",
        "userId": "XXX",
        "sn": "XXX",
        "name": "XXX",
        "inverterMeterModel": 1,
        "collectorsn": "XXX",
        "collectorId": "XXX",
        "state": 1,
        "stateExceptionFlag": 0,
        "mpptSwitch": 0,
        "collectorState": 1,
        "collectorModel": "WIFI",
        "simFlowState": -5,
        "fullHour": 3.00,
        "fullHourStr": "h",
        "currentState": "3",
        "alarmState": 1,
        "warningInfoData": 0,
        "shelfBeginTime": 1692806400000,
        "shelfEndTime": 2008425600000,
        "updateShelfEndTime": 2008425600000,
        "updateShelfEndTimeStr": "24/08/2033",
        "shelfState": "0",
        "timeZone": 1.00,
        "timeZoneStr": "UTC+01:00",
        "daylight": 0,
        "daylightSwitch": 0,
        "model": "0103",
        "productModel": "0103",
        "isS5": 0,
        "isSeparateLoad": 0,
        "isShowPowerFactor": 0,
        "isShowInternalBatteryI": 0,
        "ctrlCommand": 1,
        "inverterType": 0,
        "nationalStandards": "14",
        "nationalStandardstr": "EN50549NL",
        "inverterTemperature": 44.500,
        "inverterTemperatureUnit": "℃",
        "inverterTemperature2": 0.0,
        "inverterTemperatureUnit2": "℃",
        "temp": 150.0,
        "tempName": "IGBT  Temperature",
        "stationName": "XXX",
        "stationType": 0,
        "stationTypeNew": 0,
        "epmType": 0,
        "synchronizationType": 0,
        "gridSwitch1": 0,
        "sno": "16FA4D",
        "money": "EUR",
        "stationId": "XXX",
        "version": "800037",
        "version2": "150000",
        "acOutputType": 0,
        "dcInputtype": 0,
        "rs485ComAddr": "101",
        "dataTimestamp": "1752845333835",
[... 624 lines cut ...]
        "readHisEnergyTimeOut": "300000",
        "isSupportEpmIn": 0,
        "epmInAddr": 0,
        "afciSupportedSec": 0,
        "parallelValid": "0",
        "buckUpShowTips": 0.0,
        "wrongVersion": 0,
        "oldEnergyMachine": 0,
        "lcdHasScreen": 0,
        "afciSAT": 6,
        "outputType": 0,
        "backup2PowerOriginal": 0.0,
        "psum": 0.000,
        "psumCal": 0.810,
        "bypassLoadPowerOriginal": 0.0,
        "reactivePowerStr": "Var",
        "apparentPowerStr": "VA",
        "acCoupledPowerOriginal": 0.0,
        "familyLoadPowerPec": "1",
        "acCoupledPowerStr": "kW",
        "dcPacStr": "kW",
        "psumStr": "kW",
        "psumCalStr": "kW"
    }
}`

TCW82, 2025-07-18:
I am wondering how smart the person on the other side is at Solis:
email histrory-> TLDR on the 17th they send me an API document, on the 18th the want me to sign an NDA for the publicly available API document???

> On Thu, 17 Jul 2025, 12:55 Solis Service - Europe, <[euservice@solisinverters.com](mailto:euservice@solisinverters.com)> wrote:
> Hoi ,
> 
> Firstly,  _**refer to our most updated API document for your reference.**_ If you encounter any issues after following that, please proceed with the steps below to rectify the API issue.
> 
> 1. Test your code using Postman API.
> 2. Kindly refer to our demo code, which is in JAVA and uses the Postman API exclusively.   
> 
>               [DEMO CODE IN JAVA](https://ginlong-product.oss-cn-shanghai.aliyuncs.com/templet/Authorization.java)
> 
> After following the above steps, if you still face issues, please provide us with the following details to help rectify your issue:
> 
> Please collect the complete request package, including request header, request body, and request frequency.

### My response:

> Aan Do, 17 Jul om 1:33 PM , Theo <[theo.moons@gmail.com](mailto:theo.moons@gmail.com)> schreef:
> Hello ,
> 
> Your response is not what i was expecting. Perhaps it would be a good movie to review the previous call I have logged.  That would be  Solis Ticket - 582735 - [#582735] .
> 
> I know they seem to be about a different inverter, ik have two, both are not working with home assistant anymore since July 12th. 
> 
> At this moment I do not know what postman api is, or how to use it. And if I am really honest, all I want is my home assistant integration back online and operational. 
> 
> 
> Looking forward to hearing from you,
> 
> Cheers, 

### Their reply:

> On Fri, 18 Jul 2025, 10:48 Solis Service - Europe, <[euservice@solisinverters.com](mailto:euservice@solisinverters.com)> wrote:
> Hoi ,
> 
> please send us the signed NDA form that we attached below . Once we receive the signed NDA we will provide you with the new control document.
>

TCW82, 2025-07-18:
I am wondering how smart the person on the other side is at Solis:
email history-> TLDR on the 17th they send me an API document, it is attached to the mail in which they refer to it on the 17th. On the 18th the want me to sign an NDA for the publicly available API document, which they also already sent to me? 

> On Thu, 17 Jul 2025, 12:55 Solis Service - Europe, <[euservice@solisinverters.com](mailto:euservice@solisinverters.com)> wrote:
> Hoi ,
>
> Firstly,  _**refer to our most updated API document for your [...]

### My response:

> Aan Do, 17 Jul om 1:33 PM ,  <[(mailto:)> schreef:
> Hello ,
> 
> Your response is not what i was expecting. Perhaps it would be a good movie to review the previous call I have logged.  That would be  Solis Ticket - 582735 - [#582735] .
> 
> I know they seem to be about a different inverter, i have two, both are not working with home assistant anymore since July 12th. 
> 
> At this moment I do not know what postman api is, or how to use it. And if I am really honest, all I want is my home assistant integration back online and operational. 
> 
> 
> Looking forward to hearing from you,
> 
> Cheers, 

### Their reply:

> On Fri, 18 Jul 2025, 10:48 Solis Service - Europe, <[euservice@solisinverters.com](mailto:euservice@solisinverters.com)> wrote:
> Hoi ,
> 
> please send us the signed NDA form that we attached below . Once we receive the signed NDA we will provide you with the new control document.
>

TCW82, 2025-07-18:
> I had the same experience but I managed to get past the first line of
> support and now talking to someone who knows more about the API. We have
> all the information we need to fix the issue in the integration I think. I
> made a PR to fix the HMI issue here and when I apply the fix to my home
> assistant integration everything is working as expected.

absolute LEGEND ! I hope it works.

plebcity, 2025-07-18:
> [@plebcity](https://github.com/plebcity) mine is hmiVersionAll": "7800", so in between yours and the one by [@TCW82](https://github.com/TCW82) , will this work after your code change too?

Yes for now I've set it to 4200+ 
Still wondering if this is the correct property to determine if an inverter is controllable or not but for now it's the same as it used to be.

plebcity, 2025-07-18:
> Is it a “simple” code change I (we) can change in a file of our current ‘custom-components’ to help you test and make it work again?
> 
> EDIT: forget my stupid question, just noticed [#483](https://github.com/hultenvp/solis-sensor/pull/483) and it are a lot of changes in 7 files 😜

No worries, yes it is a bigger change than just changing a value. You can test the change if you know how to load a custom component.

TCW82, 2025-07-18:
I will just sit here and wait .. 🤣

imc67, 2025-07-19:
> No worries, yes it is a bigger change than just changing a value. You can test the change if you know how to load a custom component.

YES!! 

1. I copied the new files into the solis custom-components folder 
2. Restart HA
3. Sensors and info came in
4. Setup Control API credentials
5. Restart HA
6. after some minutes all data came in and the controls are there AND working

@plebcity thanks you very very much!!! Enorm bedankt Koen voor je aanpassingen, echt blij mee!

TCW82, 2025-07-19:
Nu nog voor de rest van ons simpele zielen.... (We mere mortals also look
forward to the implementation of this fix)

sjanssen15, 2025-07-25:
Is there any update on this issue? The integration isnt working for 2 weeks now for me. I saw there was a PR that could solve this? Of am I wrong

TCW82, 2025-07-26:
> Is there any update on this issue? The integration isnt working for 2 weeks now for me. I saw there was a PR that could solve this? Or am I wrong

Maybe the pr takes a little longer due to the summer holiday periode ?

thvd, 2025-07-29:
> Has anyone tried the new release? Did it address this issue?

Yes it's loading again fine.

jmason, 2025-07-29:
hooray, closing this bug report 🎉

closed 2025-07-29 by jmason