hultenvp/solis-sensor #492: Inverter control not available in 3.13

TCW82, 2025-08-10:
**Describe the bug**
after update tot 3.13 only 1 of two inverters has control options (switch off /on, limit output). 

**Config**
Will add later, issu looks a lot like the original issue, although I have already gone thru all the motions of adding an additional plant and so on. 

**Versions**
HA Version an Deployment: latest as of 10 aug 2025
HACS version latest as of 10 aug 2025
Integration version 3.13( although it still shows 3.12 in home assistant??) 

I have no logs to add, the options are not there in the home assistant environment. I can control the inverter (both!) via solis cloud.

fboundy, 2025-08-13:
Can you add the following to `configuration.yaml' so that logs are generated:

    logger:
      default: error
      logs:
        # log level for a custom integration
        custom_components.solis: debug

TCW82, 2025-08-13:
> Can you add the following to `configuration.yaml`so that logs are generated:
> 
> ```
> logger:
>   default: error
>   logs:
>     # log level for a custom integration
>     custom_components.solis: debug
> ```

Done, what is the next step. I am not a real specialist.

TCW82, 2025-08-13:
> If you restart HA and then go to Settings\System\Logs you should be able
> to download the log and post it on here.

[36m2025-08-13 17:24:52.482 DEBUG (MainThread) [custom_components.solis.select] >>> Value: 190[0m
[36m2025-08-13 17:24:52.482 DEBUG (MainThread) [custom_components.solis.select] >>> Lookup: ON[0m
[36m2025-08-13 17:24:52.482 DEBUG (MainThread) [custom_components.solis.select] >>> Last updated: 2025-08-13 17:24:52.472367[0m
[36m2025-08-13 17:24:52.482 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:04:15.990643, at 2025-08-13 15:29:08.473240+00:00[0m
[31m2025-08-13 17:25:38.415 ERROR (MainThread) [aioesphomeapi.connection] s5-sonair-sk-c @ 192.168.1.22: disconnect request failed
Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 878, in aioesphomeapi.connection.APIConnection.send_messages_await_response_complex
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 1096, in aioesphomeapi.connection.APIConnection.disconnect
  File "aioesphomeapi/connection.py", line 896, in send_message_await_response
  File "aioesphomeapi/connection.py", line 882, in send_messages_await_response_complex
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for DisconnectResponse after 10.0s[0m
[31m2025-08-13 17:26:53.689 ERROR (MainThread) [aioesphomeapi.connection] s5-sonair-sk-c @ 192.168.1.22: disconnect request failed
Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 878, in aioesphomeapi.connection.APIConnection.send_messages_await_response_complex
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 1096, in aioesphomeapi.connection.APIConnection.disconnect
  File "aioesphomeapi/connection.py", line 896, in send_message_await_response
  File "aioesphomeapi/connection.py", line 882, in send_messages_await_response_complex
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for DisconnectResponse after 10.0s[0m
[36m2025-08-13 17:28:06.072 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110AA119B300027[0m
[36m2025-08-13 17:28:08.297 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today[0m
[36m2025-08-13 17:28:08.299 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching control data for SN:110AA119B300027[0m
[31m2025-08-13 17:28:08.914 ERROR (MainThread) [aioesphomeapi.connection] s5-sonair-sk-c @ 192.168.1.22: disconnect request failed
Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 878, in aioesphomeapi.connection.APIConnection.send_messages_await_response_complex
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "aioesphomeapi/connection.py", line 1096, in aioesphomeapi.connection.APIConnection.disconnect
  File "aioesphomeapi/connection.py", line 896, in send_message_await_response
  File "aioesphomeapi/connection.py", line 882, in send_messages_await_response_complex
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for DisconnectResponse after 10.0s[0m
[36m2025-08-13 17:28:09.051 DEBUG (MainThread) [custom_components.solis.soliscloud_api]     cid: 15    - 110[0m
[32m2025-08-13 17:28:10.485 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 157   - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:11.485 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 158   - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:12.586 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 160   - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[36m2025-08-13 17:28:13.283 DEBUG (MainThread) [custom_components.solis.soliscloud_api]     cid: 230   - 0[0m
[32m2025-08-13 17:28:14.259 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 636   - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:15.254 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 696   - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[36m2025-08-13 17:28:15.562 DEBUG (MainThread) [custom_components.solis.soliscloud_api]     cid: 5161  - 190[0m
[32m2025-08-13 17:28:16.803 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 5928  - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:17.793 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 5948  - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:18.792 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 5946  - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:19.574 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 5965  - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:21.080 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 5967  - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[32m2025-08-13 17:28:22.079 INFO (MainThread) [custom_components.solis.soliscloud_api]     cid: 5964  - /v2/api/atRead responded with error: B0600:采集器返回数据异常，请稍后再试[0m
[... cut ...]
  File "aioesphomeapi/connection.py", line 896, in send_message_await_response
  File "aioesphomeapi/connection.py", line 882, in send_messages_await_response_complex
aioesphomeapi.core.TimeoutAPIError: Timeout waiting for DisconnectResponse after 10.0s[0m
[36m2025-08-13 17:43:06.260 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110AA119B300027[0m
[36m2025-08-13 17:43:08.502 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today[0m
[36m2025-08-13

closed 2025-10-30 by TCW82