hultenvp/solis-sensor #496: Write defaults for refresh rates in config when not present

hultenvp, 2025-08-30:
**Describe the bug**
Currently as workaround there's a try/except clause around getting the refresh rates from config.
When not found it just uses defaults.
Better would be to automatically update the config when values are missing

**Versions**
HA Version an deployment
HACS version
Integration version: 3.13.1

`    # Initialize the Ginlong data service.`
`    refresh_ok = 300`
`    refresh_error = 60`
`    try:`
`        # Fixme:` 
`        refresh_ok = config[CONF_REFRESH_OK]`
`        refresh_error = config[CONF_REFRESH_NOK]`
`    except KeyError:`
`        pass`

hultenvp, 2025-08-30:
@corvus2606 FYI. Your config changes did did not work when restarting HA without manual resubmission of config.

dpgh947, 2025-08-30:
Is this why changes to refresh_ok in the new config fields are not sticking? I requested a 60 second update interval on my inverter some time ago, and have been editing service.py to implement this after any integration updates. Now I can't set this at all. I change the field from 300 to 60, press submit and it reverts to 300.

hultenvp, 2025-08-30:
No if that happens then there's an issue. It should still be possible to set intervals, the only thing I added on corvus' change is that the integration does not crash when the config does not exist.