hultenvp/solis-sensor #497: Changes to the new config refresh interval fields do not stick

dpgh947, 2025-08-31:
**Describe the bug**
A clear and concise description of what the bug is.

I have been manually editing service.py to set the SCHEDULE_OK interval to 60 since I requested a 60 second refresh on my inverter from Solis. The latest update exposes these refresh fields as config options, but any attempt to change them is ignored. If I change the main refresh from 300 to 60 and press submit, it immediately reverts to 300. 

I also notice that the integration page still says the version is 3.12.1, the github page says 3.13.1. I have forced a redownload through HACS which says it is 3.13.1, but the integration page still says 3.12.1

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment 2025.8.3, HAOS 16.1
HACS version 2.0.5
Integration version 3.13.1 ????

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

I have enabled debug as above, I see debug messages during startup so I know it is working, but using the config ui to attempt the change produces no messages.

**Screenshots**
If applicable, add screenshots to help explain your problem.

dpgh947, 2025-08-31:
Well... I have deleted and re-installed the integration. This caused me a massive headache, because all the entity names got recreated with an extra 
```
"_solis_"
```
inserted in the names, I had to edit them all...

Anyway, when adding the inverter the new refresh fields were there on the initial config screen too. I set them both to 60. I can see from looking in .storage/core.config_entries that this is being saved correctly - 
```
refresh_nok":60,"refresh_ok":60
```
but if I then go in to the ui config again, the first refresh field (the "ok" one) is still coming up with 300 in it.

I can also see from the debug log that it is updating every minute, but there seem to be two update processes going on, here is the debug log from the startup, all the "xxxxxx" names are the same single inverter name, I do not have two. It keeps saying next update in 1 minute, it does that, then says next update in a few seconds, does that, then next update in 1 minute again...
```
2025-08-31 10:24:52.240 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2025-08-31 10:24:52.240 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['xxxxxxxxxxxxxxxxxx']
2025-08-31 10:24:52.240 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: xxxxxxxxxxxxxxxxxx
2025-08-31 10:24:54.374 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-08-31 10:24:54.374 DEBUG (MainThread) [custom_components.solis.soliscloud_api] HMI firmware version (4000) <4B00 for Inverter SN xxxxxxxxxxxxxxxxxx 
2025-08-31 10:24:54.374 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Valid inverters: ['xxxxxxxxxxxxxxxxxx']
2025-08-31 10:24:54.374 INFO (MainThread) [custom_components.solis.soliscloud_api] No control password set; control mode disabled
2025-08-31 10:24:54.374 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: xxxxxxxxxxxxxxxxxx
2025-08-31 10:24:56.496 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-08-31 10:24:56.507 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:01:00, at 2025-08-31 09:25:56.507745+00:00
2025-08-31 10:25:56.509 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: xxxxxxxxxxxxxxxxxx
2025-08-31 10:25:58.733 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-08-31 10:25:58.739 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:00:10.265569, at 2025-08-31 09:26:09.004795+00:00
2025-08-31 10:26:09.005 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: xxxxxxxxxxxxxxxxxx
2025-08-31 10:26:11.127 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-08-31 10:26:11.128 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:01:00, at 2025-08-31 09:27:11.128338+00:00
2025-08-31 10:27:11.129 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: xxxxxxxxxxxxxxxxxx
2025-08-31 10:27:13.367 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-08-31 10:27:13.368 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:00:07.632342, at 2025-08-31 09:27:21.001236+00:00
2025-08-31 10:27:21.002 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: xxxxxxxxxxxxxxxxxx
2025-08-31 10:27:23.147 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2025-08-31 10:27:23.147 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:01:00, at 2025-08-31 09:28:23.147835+00:00
```

corvus2606, 2025-08-31:
> I have tested changing the refresh to different values, and even though it always resets to 300 in the ui, I can see the different values being saved in core.config_entries. So the change is being saved and used, but the ui is never showing it, it always comes up filled with 300 and reverts to 300 when you submit.

Apologies, I've been working away, so haven't been able to keep on top of this.  unfortunately the config update flow is a bit weird and once you submit, it always shows the default values.  

I'm relatively new to this part of ha extensions so I've only added the fields to the flow and everything else is handled the same as before.  I tried getting it to read from the config and set that as the default, but ran out of time.  I'll see if I can figure it out

dpgh947, 2025-08-31:
Ok. What about the multiple update calls in the log I posted? It might always have done this, I only spotted it now when debug was on, but it doesn't look right.