hultenvp/solis-sensor #505: KeyError: 'portal_version'

krizzziz, 2025-09-27:
I just setup my config, I have put in all the info a regular and the system accepted the userlogin, API and PW without problems. But then when it needs to setup the sensors, the system is giving me this error:

`Logger: homeassistant.config_entries
Bron: config_entries.py:751
Eerst voorgekomen: 11:38:10 (1 gebeurtenis)
Laatst gelogd: 11:38:10

Error setting up entry Station 2A7815 for solis
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 751, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis/__init__.py", line 86, in async_setup_entry
    portal_version = config[CONF_PORTAL_VERSION]
                     ~~~~~~^^^^^^^^^^^^^^^^^^^^^
KeyError: 'portal_version'`

What to do?

hultenvp, 2025-09-28:
Hmmz, not sure how that slipped through, but yes that's for sure semi-removed ;-)

hultenvp, 2025-09-28:
@krizzziz @matthiaz : Just pushed v4.0.1. That should fix your issue.
I'm having troubles with my setup, so cannot fully test (not able to create a fresh config without breaking my own setup)
Can you please let me know if it works or not? Just reopen this issue if you still encounter issues.