hultenvp/solis-sensor #506: Cannot login with provided URL and credentials

rfcdejong, 2025-09-27:
I have a working integration with my soliscloud, but added a second inverter with a new datalogger.

However when i try to add a new integration item, filling a new unique name, the url  https://www.soliscloud.com:13333, username, the apikey id and secret and the station id. I keep getting 

"Cannot login with provided URL and credentials" 

I have tried my username as username and email. Tripple checked api key id, secret and station id. Updated to the latest version and all...

What to do? Turned on debug logging.

Best error I can find

Logger: custom_components.solis.soliscloud_api
Bron: custom_components/solis/soliscloud_api.py:344
integratie: Solis Inverter ([documentatie](https://github.com/hultenvp/solis-sensor/), [problemen](https://github.com/hultenvp/solis-sensor/issues))
Eerst voorgekomen: 22:07:41 (3 gebeurtenissen)
Laatst gelogd: 22:20:18

No inverters found

hultenvp, 2025-09-28:
Also, not sure if you use the control API, but try starting with control API off.

I'll release a 4.0.1 later today. It fixes a bug for new configurations, but that bug gave a keyerror, not a "no inverters found". Still might be worth giving a try.

rfcdejong, 2025-09-30:
Seems I was using the wrong url. Never the less the error is misleading.

Needed to copy the station ID from 
https://www.soliscloud.com/station/stationDetails/generalSituation/xxxxxxxx

I was using this one, which is wrong
https://www.soliscloud.com/station/inverter/inverterdetail?id=xxxxxxx

hultenvp, 2025-10-01:
Sorry, my link was pointing to a development branch which was closed already. Hence the 404.

I'll see if we can improve the error logs, but good it is solved.

closed 2025-10-01 by hultenvp